litex-hub/linux-on-litex-rocket #2: Add a simplified diagram of the SoC in the README?

enjoy-digital, 2020-12-18:
Adding a simplified / minimal diagram of the SoC to the README would I think provide a quick and better overview of the aim of the project and what's built with it.

Here is a first attempt I just did:
[image]

@gsomlo: If you could be interested, happy to review / do changes to it and create a PR to the README. I also don't want to change the style of the project or transform it, so if you have preference for a full text README or wants to create your own diagram, feel to reject this.

gsomlo, 2020-12-18:
@enjoy-digital: diagram looks great, thanks for creating it, and looking forward to the PR!

One thought: right now the DMA bus is essentially a point-to-point link between Rocket and LiteSDCard, but would you lay it out slightly differently knowing that we might one day (hopefully soon) add LiteSATA to the design, which will also need to be connected to the DMA bus? Making sure that addition will require minimal (visual) re-arrangement of the existing components might be worth a few seconds' thought :) 

Finally, not sure if we want to add IRQs to the diagram or not (probably not, but just thinking out loud).

enjoy-digital, 2020-12-18:
@gsomlo: Thanks for the feedback, is it better like this? (rearranged the DMA + added IRQs):

[image]

Also, do you want the LiteX-Hub logo on it ? (feel free to say no, I just added it here to show you)

gsomlo, 2020-12-18:
@enjoy-digital: looks good, thanks! Also, LiteX-Hub logo is fine with me (although, visually, I'd probably place it somewhere underneath `/dev/ttyUSBX` on the left (probably aligned with the bottom of the rest of the image) where it wouldn't have to stick out vertically. It's a minor thing, perfectly OK where it is now if "bottom-right" is sort-of a standard for logo placement...

enjoy-digital, 2020-12-18:
PR https://github.com/litex-hub/linux-on-litex-rocket/pull/3 created with:

[image]

I'll also share with you the source of this diagram.

gsomlo, 2020-12-18:
Thanks again, looks really good. Depending on what the source looks like, we might want to make that part of the repo itself -- what do you think?

gsomlo, 2020-12-18:
resolved by merging PR #3 -- Thanks again @enjoy-digital for putting it together!

closed 2020-12-18 by gsomlo