litex-hub/linux-on-litex-rocket #4: Set up on Ubuntu might be incomplete

tommythorn, 2020-12-21:
My x64 host is running Ubuntu 21.04 (Hirsute Hippo), fully up to date.  I tried to map the step 1 to equivalent Ubuntu repos (I would need to this on a clean installation to complete this).

```
$ sudo apt install libjson-c-dev libcrypt-openssl-bignum-perl libevent-dev libmpc-dev libmpfr-dev python3-dev
```

but step `python3 ./litex_setup.py init install --user` fails with
```
$ python3 ./litex_setup.py init install --user
[checking litex_setup.py]...
[installing migen]...
Traceback (most recent call last):
  File "/home/tommy/projects/migen/setup.py", line 4, in <module>
    from setuptools import setup
ModuleNotFoundError: No module named 'setuptools'
Traceback (most recent call last):
  File "/home/tommy/projects/linux-on-litex-rocket/./litex_setup.py", line 168, in <module>
    subprocess.check_call("python3 setup.py develop --user", shell=True)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'python3 setup.py develop --user' returned non-zero exit status 1.
```

gsomlo, 2020-12-21:
On Fedora, `python3-setuptools` is a dependency of `python3-devel`, so you get it "for free". With commit 568513a I listed it explicitly, so hopefully that takes care of it.

closed 2020-12-21 by gsomlo

tommythorn, 2021-01-04:
Getting beyond this point, the instructions seems to assume you have a "litex-boards" directory.  Where is that supposed to come from?

gsomlo, 2021-01-05:
> Getting beyond this point, the instructions seems to assume you have a "litex-boards" directory. Where is that supposed to come from?

From [`litex_setup.py`](https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L43) (Step 2 of the prerequisites).