litex-hub/linux-on-litex-rocket #5: just scrolling LEDs

monniaux, 2021-01-12:
On Nexys-A7, if the MODE jumper is set to JTAG:

Programming the board with openocd -f litex-boards/litex_boards/prog/openocd_xc7_ft2232.cfg -c 'transport select jtag; init; pld load 0  linux-on-litex-rocket/nexys4ddr.bit ; exit' just results in the board scrolling its LEDs without booting anything.

Is there anything else to set?

gsomlo, 2021-01-12:
You're talking about the "programming mode" jumper (#9 in the linked
image: https://reference.digilentinc.com/_media/reference/programmable-logic/nexys-a7/nexys-a7-callout.png

The other potentially relevant jumper is "external configuration
(SD/USB)" (#23 in the image above). That one should be set to "USB".

gsomlo, 2021-01-14:
@monniaux: Any luck? I only have the "classic" nexys4ddr board, but word on the street is that the nexys-a7 should be compatible. Not sure if we're tickling any small-but-relevant differences between the two versions of the board, but it would be nice to know if that's the case...

monniaux, 2021-01-14:
Setting switches: MODE: USB/SD, JP2: USB, SW1: 1, the system boots. Thank you. I am going to run some tests !

closed 2021-01-14 by monniaux