litex-hub/linux-on-litex-rocket #6: hangs and unstable cycle counts

monniaux, 2021-01-14:
Running some test code. Sometimes the programs (which work perfectly under simulators; including GNU make, a version of the OCaml runtime…) seem to hang (control-C exits though).

Also, when they don't hang, they seem to be randomly affected by some delays leading to high cycle counts.

Could there be that sometimes accesses to the SD card fail and hang programs?

gsomlo, 2021-01-14:
On Thu, Jan 14, 2021 at 12:36:56PM -0800, David Monniaux wrote:
> Could there be that sometimes accesses to the SD card fail and hang
> programs?

Yes. SD card support (mainly the Linux litex_mmc driver, but also to
an extent the LiteSDCard gateware itself) is still a work in progress. 

At the moment, at least when using 64-bit Linux and LiteX + Rocket,
there are data and DMA timeouts, and only single-block transfers are
enabled in the Linux driver.

I'm in the process of tracking down the source of those bottlenecks,
so things should improve, hopefully soon.

enjoy-digital, 2021-01-16:
Thanks @monniaux, would you mind sharing your test? We could use it as an additionnal stress test while looking at the other SDCard issues.

gsomlo, 2021-02-22:
see also https://github.com/enjoy-digital/litesdcard/issues/22

gsomlo, 2021-04-22:
LiteSDCard is now working fine, so I'm going to close this.

@monniaux -- please feel free to (re-)open this or a new issue if you're still experiencing any trouble.

closed 2021-04-22 by gsomlo