litex-hub/linux-on-litex-rocket #8: Resources on creating a proper device file tree

developandplay, 2021-03-28:
Thanks for your help on #7.
Now that I sucessfully flashed the svf (_with one warning: Max frequency for clock '$glbnet$main_clkout0': 11.27 MHz (FAIL at 50.00 MHz)_) on my board I was following the next steps.
Therefore I was wondering whether you have some more resources on how you created the dft for your boards.

gsomlo, 2021-03-28:
> Therefore I was wondering whether you have some more resources on how you created the dft for your boards.

Do you mean `*.dts` (device tree source files)? If so, it started out as a combination of the `.dts` generated by the RocketChip builder (see e.g. the [`DTS`](https://github.com/litex-hub/pythondata-cpu-rocket/blob/master/pythondata_cpu_rocket/verilog/generated-src/freechips.rocketchip.system.LitexLinuxConfig.dts) generated for the 64-bit-wide `linux` variant) and the MMIO (a.k.a. CSR) addresses generated by LiteX for the SoC peripherals: if you build with `--csr-csv ./csr.csv` or better yet, with both that and `--csr-json ./csr.json`, you can then run `litex_json2dts.py --initrd-start 8 ./csr.json` and use ***some*** of its output.

In your case, I'd start with [nexys4ddr.dts](https://github.com/litex-hub/linux-on-litex-rocket/blob/master/conf/nexys4ddr.dts) (since it also uses the 64-bit-wide Rocket CPU variant):
    - update [here](https://github.com/litex-hub/linux-on-litex-rocket/blob/master/conf/nexys4ddr.dts#L48) with the actual size of the RAM on your board (look for `memory_region,main_ram,0x80000000,<SIZE>,cached`, convert `<SIZE>` to hex, and you're good to go;
    - replace [this section](https://github.com/litex-hub/linux-on-litex-rocket/blob/master/conf/nexys4ddr.dts#L90-L115) with the matching portion of output from the `litex_json2dts.py`, and then ***INCREMENT*** each IRQ number in the pasted content by 1 (i.e., the IRQ for the UART changes from 0 to 1, and that of LiteETH from 2 to 3)!

In the near future, it is one of my goals for this project to "teach" `litex_json2dts.py` to generate the `*.dts` file automatically, and get rid of the individual board-specific examples in [`./conf`](https://github.com/litex-hub/linux-on-litex-rocket/tree/master/conf). But, in the mean time, doing something along the lines of what's listed above should take care of it.

developandplay, 2021-03-28:
Oh yes sorry for the type that's exactly what I meant.
Thanks for the help!

developandplay, 2021-03-29:
Which is the _Max frequency for clock '$glbnet$main_clkout0':_ that you are getting for your (ECP5) boards?
I fail to upload the the boot.bin to the ULX3S and have read that this might be due to too low clock frequency. https://github.com/enjoy-digital/litex/issues/773
For the vexrisc cpu the frequency the _Max frequency_ is above 50Mhz so it's all fine...

gsomlo, 2021-03-29:
On Mon, Mar 29, 2021 at 10:36:05AM -0700, developandplay wrote:
> Which is the Max frequency for clock '$glbnet$main_clkout0': that you are
> getting for your (ECP5) boards?

Generally I get somewhere from 19MHz to 25MHz from nextpnr. It's
supposedly ultra-conservative, since the resulting design mostly
works OK at 50MHz regardless.

developandplay, 2021-06-07:
In case anyone else runs into this issue.
Make sure to run `lxterm /dev/ttyUSB0 --kernel boot.bin --speed 115200 --kernel-adr 0x80000000` as the default address is 0x40000000 which of course won't work with rocket.
With this the upload over serial will work. However it takes about 1,5h.

developandplay, 2021-06-07:
Seems like my dts still wasn't set up properly however since BBL is not starting yet.

developandplay, 2021-06-08:
Turns out my riscv-pk was not up to date.
Now I finally get bbl to start on the arty.
However it gets stuck before the login screen.
```
bbl loader
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 5.12.0-rc4-157053-g2669e46c615f (martin@martin-ThinkPad-T14-Gen-1) (riscv64-unknown-linux-gnu-gcc (GCC) 9.2.0, GNU ld (GNU Binutils) 2.34) #1 Mon Mar 29 15:25:44 CEST 2021
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Machine model: freechips,rocketchip-unknown
[    0.000000] earlycon: sbi0 at I/O port 0x0 (options '')
[    0.000000] printk: bootconsole [sbi0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x000000008fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x000000008fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x000000008fffffff]
[    0.000000] On node 0 totalpages: 65024
[    0.000000]   DMA32 zone: 1016 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 65024 pages, LIFO batch:15
[    0.000000] SBI specification v0.1 detected
[    0.000000] software IO TLB: mapped [mem 0x000000008fbfa000-0x000000008fbfa800] (0MB)
[    0.000000] riscv: ISA extensions acdfim
[    0.000000] riscv: ELF capabilities acdfim
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64008
[    0.000000] Kernel command line: earlycon=sbi console=liteuart swiotlb=noforce
[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 242252K/260096K available (3729K kernel code, 4061K rwdata, 2048K rodata, 2808K init, 270K bss, 17844K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 64 local interrupts mapped
[    0.000000] plic: interrupt-controller@c000000: mapped 4 interrupts with 1 handlers for 2 contexts.
[    0.000000] random: get_random_bytes called from start_kernel+0x3a0/0x558 with crng_init=0
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[    0.000023] sched_clock: 64 bits at 1000kHz, resolution 1000ns, wraps every 2199023255500ns
[    0.006260] Console: colour dummy device 128x32
[    0.008372] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.013330] pid_max: default: 32768 minimum: 301
[    0.017647] LSM: Security Framework initializing
[... 11 lines cut ...]
[    0.356830] TCP established hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.360970] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.364923] TCP: Hash tables configured (established 2048 bind 2048)
[    0.369107] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.372598] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.377886] NET: Registered protocol family 1
[    0.937063] workingset: timestamp_bits=46 max_order=16 bucket_order=0
[    1.181944] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
[    1.188538] LiteX SoC Controller driver initialized: subreg:4, align:4
[    2.088147] loop: module loaded
[    2.264756] libphy: Fixed MDIO Bus: probed
[    2.276462] liteeth 12001000.mac eth0: irq 2, mapped at ffffffd004003000
[    2.300900] NET: Registered protocol family 10
[    2.318173] Segment Routing with IPv6
[    2.320626] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    2.333819] NET: Registered protocol family 17
[    2.338181] Warning: unable to open an initial console.
[    2.382536] Freeing unused kernel memory: 2808K
[    2.384950] Run /init as init process
[    2.386312]   with arguments:
[    2.387787]     /init
[    2.388917]   with environment:
[    2.390481]     HOME=/
[    2.391653]     TERM=linux
```

developandplay, 2021-06-09:
Success on the arty #13 !!!

developandplay, 2021-06-09:
I will try to see if I can get linux to boot on the ulx3s next.

developandplay, 2021-06-25:
Closing this as #14 and #12 are not dependent on it.

closed 2021-06-25 by developandplay

lapnd, 2022-02-12:
Hi @developandplay 
I'm also getting hang at this step when trying on DE2-115 board
```sh
[   31.277582] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   31.476334] NET: Registered PF_PACKET protocol family
[   32.024800] Warning: unable to open an initial console.
[   32.817778] Freeing unused kernel image (initmem) memory: 2700K
[   32.862982] Run /init as init process
[   32.867512]   with arguments:
[   32.876518]     /init
[   32.882666]   with environment:
[   32.886522]     HOME=/
[   32.892520]     TERM=linux
```
Can you share, what did you had done to fix this issue?

lapnd, 2022-02-12:
Rebuild kernel with default option + Enable SBI v1 make it work for me as https://github.com/litex-hub/linux-on-litex-rocket/issues/21