litex-hub/linux-on-litex-rocket #9: Design doesn't fit on OrangeCrab FPGA

cjearls, 2021-04-17:
I'd really like to be able to run this on the OrangeCrab, which has an EPC5 LFE5U-25F. Unfortunately, it uses 20,360 TRELLIS_SLICE blocks, but the board only has 12,144 of them. Are there any optimizations that could be made to fit this onto a smaller FPGA? Is the Rocket core the smallest possible that is capable of running Linux? Could more functionality be optionally removed from it and emulated in BBL?

gsomlo, 2021-04-17:
Hi @cjearls -- There is a "standard" rocket variant in Litex which comes without an MMU, and therefore *can't* run Linux, but might fit within fewer TRELLIS_SLICEs and run bare-metal programs. I don't think it's fewer *enough* slices for a 25k ECP5 though, unfortunately.

As it is, the "linux"-capable variant *barely* fits into a 45k ECP5 (on the Versa board) -- including litedram and liteeth, which is useful. There'd clearly be no room left for e.g. sdcard gateware, if anyone ever figured out how to hook up an sdcard reader to the Versa.

Not sure about other 64-bit RISC-V options (besides Rocket) and Linux, it might be worth a few minutes to investigate, if only to rule out the idea...

With a 25K ECP5 you might be forced to stick to 32-bit CPUs.

Another thing worth a few minutes of research would be to figure out how much better placement is with the vendor toolchain (Diamond) as opposed to yosys/trellis/nextpnr. I don't expect it to be better by enough to fit a rocket-capable litex soc into 25K slices, but since I'm grasping at straws here, I figured I'd throw it in...

Sorry I don't have a better answer...

cjearls, 2021-04-21:
Thanks for your help!

closed 2021-04-21 by cjearls