litex-hub/linux-on-litex-rocket #12: Booting into linux on simulation

developandplay, 2021-06-08:
> At the moment, I'm not aware of a good way to tell the firmware/BIOS to simply jump to the first RAM address and start executing the side-loaded boot.bin which is now located there. Although that should be an easy fix.

For linux-on-litex-blackparrot we are setting the `ROM_BOOT_ADDRESS`  in `litex_sim.py` to `0x80000000`.
However even with that change I didn't manage to boot into linux with rocket.
Do you have exact steps that you followed to accomplish that?
Using a method other than side loading would be fine as well of course.

I just want to make sure that I'm having a reproducible boot into linux in simulation before I further attempt it on my ulx3s and arty7.
For the ulx3s so far I'm reliably getting into the bios. On the arty7 I encounter some [memory errors](https://github.com/enjoy-digital/litex/issues/933#issue-914058679) with newer versions of litex.

gsomlo, 2021-06-08:
Last time I tried this was a very long time ago. Make sure you have `libevent-devel` and `json-c-devel` installed. The command line where you load `boot.bin` over tftp should be:

```
litex/litex/tools/litex_sim.py \
  --threads 8 --opt-level Ofast \
  --with-ethernet --with-sdram \
  --cpu-type rocket --cpu-variant linux
```

However, simulating 64-bit rocket running the full tftp download sequence is about an order of magnitude less entertaining than watching paint dry. So, I remember replacing `--with-sdram` with `--ram-init boot.bin`, to bypass tftp. Hope it all still works...

developandplay, 2021-06-08:
Hm I'm still not getting it to start unfortunately.
This is the csr.csv that Litex generates for me:
```
#--------------------------------------------------------------------------------
# Auto-generated by Migen (3ffd64c) & LiteX (bd146351) on 2021-06-08 21:07:56
#--------------------------------------------------------------------------------
csr_base,ctrl,0x12000000,,
csr_base,ethmac,0x12000800,,
csr_base,ethphy,0x12001000,,
csr_base,identifier_mem,0x12001800,,
csr_base,timer0,0x12002000,,
csr_base,uart,0x12002800,,
csr_register,ctrl_reset,0x12000000,1,rw
csr_register,ctrl_scratch,0x12000004,1,rw
csr_register,ctrl_bus_errors,0x12000008,1,ro
csr_register,ethmac_sram_writer_slot,0x12000800,1,ro
csr_register,ethmac_sram_writer_length,0x12000804,1,ro
csr_register,ethmac_sram_writer_errors,0x12000808,1,ro
csr_register,ethmac_sram_writer_ev_status,0x1200080c,1,ro
csr_register,ethmac_sram_writer_ev_pending,0x12000810,1,ro
csr_register,ethmac_sram_writer_ev_enable,0x12000814,1,rw
csr_register,ethmac_sram_reader_start,0x12000818,1,rw
csr_register,ethmac_sram_reader_ready,0x1200081c,1,ro
csr_register,ethmac_sram_reader_level,0x12000820,1,ro
csr_register,ethmac_sram_reader_slot,0x12000824,1,rw
csr_register,ethmac_sram_reader_length,0x12000828,1,rw
csr_register,ethmac_sram_reader_ev_status,0x1200082c,1,ro
csr_register,ethmac_sram_reader_ev_pending,0x12000830,1,ro
csr_register,ethmac_sram_reader_ev_enable,0x12000834,1,rw
csr_register,ethmac_preamble_crc,0x12000838,1,ro
csr_register,ethphy_crg_reset,0x12001000,1,rw
csr_register,timer0_load,0x12002000,1,rw
csr_register,timer0_reload,0x12002004,1,rw
csr_register,timer0_en,0x12002008,1,rw
csr_register,timer0_update_value,0x1200200c,1,rw
csr_register,timer0_value,0x12002010,1,ro
csr_register,timer0_ev_status,0x12002014,1,ro
csr_register,timer0_ev_pending,0x12002018,1,ro
csr_register,timer0_ev_enable,0x1200201c,1,rw
csr_register,uart_rxtx,0x12002800,1,rw
csr_register,uart_txfull,0x12002804,1,ro
csr_register,uart_rxempty,0x12002808,1,ro
csr_register,uart_ev_status,0x1200280c,1,ro
csr_register,uart_ev_pending,0x12002810,1,ro
csr_register,uart_ev_enable,0x12002814,1,rw
csr_register,uart_txempty,0x12002818,1,ro
csr_register,uart_rxfull,0x1200281c,1,ro
constant,config_clock_frequency,1000000,,
constant,config_cpu_has_interrupt,None,,
constant,config_cpu_reset_addr,268435456,,
constant,config_cpu_type_rocket,None,,
constant,config_cpu_variant_linux,None,,
constant,config_cpu_human_name,rocketrv64[imac],,
constant,config_cpu_nop,nop,,
constant,config_with_build_time,None,,
constant,rom_boot_address,2147483648,,
constant,localip1,192,,
constant,localip2,168,,
constant,localip3,1,,
constant,localip4,50,,
constant,remoteip1,192,,
constant,remoteip2,168,,
constant,remoteip3,1,,
constant,remoteip4,100,,
constant,config_csr_data_width,32,,
constant,config_csr_alignment,32,,
constant,config_bus_standard,wishbone,,
constant,config_bus_data_width,32,,
constant,config_bus_address_width,32,,
constant,config_cpu_has_dma_bus,None,,
[... 143 lines cut ...]
 Migen git sha1: 3ffd64c
 LiteX git sha1: bd146351

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 1MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
MAIN-RAM:       262144KiB 

--========== Initialization ============--
Ethernet init...

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x80000000

--============= Liftoff! ===============--
```

Maybe I'm not waiting long enough but for blackparrot and vexriscv at least the `BBL loader` line gets shown pretty much instantly.

developandplay, 2021-06-24:
I revisited this issue and made some progress. Surprisingly I ran into a kernel panic.

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 23 2021 21:32:11
 BIOS CRC passed (16f2dc15)

 Migen git sha1: 3ffd64c
 LiteX git sha1: adb71bde

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 1MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
SDRAM:          65536KiB 32-bit @ 1MT/s (CL-2 CWL-2)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Cancelled

--============= Console ================--

litex> romboot
Executing booted program at 0x80000000

--============= Liftoff! ===============--
bbl loader
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
rrrrrrrrrr      vvvvvvvvvvvvvv      rrrrrrrrrr
rrrrrrrrrrrr      vvvvvvvvvv      rrrrrrrrrrrr
rrrrrrrrrrrrrr      vvvvvv      rrrrrrrrrrrrrr
rrrrrrrrrrrrrrrr      vv      rrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrr          rrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrr      rrrrrrrrrrrrrrrrrrrr
rrrrrrrrrrrrrrrrrrrrrr  rrrrrrrrrrrrrrrrrrrrrr

       INSTRUCTION SETS WANT TO BE FREE
[    0.000000] Linux version 5.13.0-rc3-173670-g9dacadb8c086 (martin@martin-ThinkPad-T14-Gen-1) (riscv64-unknown-linux-gnu-gcc (GCC) 9.2.0, GNU ld (GNU Binutils) 2.34) #3 Wed Jun 9 01:39:48 CEST 2021
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[... 72 lines cut ...]
[    2.384676]  s8 : ffffffff80600468 s9 : 0000000000000008 s10: ffffffff804000a8
[    2.386312]  s11: 0000000000000000 t3 : 0000000000001000 t4 : ffffffff80ca71c8
[    2.387921]  t5 : ffffffd004002000 t6 : 0000000000000000
[    2.389271] status: 0000000200000120 badaddr: ffffffd004001b04 cause: 0000000000000005
[    2.390956] Call Trace:
[    2.391978] [<ffffffff802420a6>] litex_soc_ctrl_probe+0x3e/0x106
[    2.394077] [<ffffffff80267f80>] platform_probe+0x3e/0x7c
[    2.395996] [<ffffffff802667ec>] really_probe+0x8e/0x26a
[    2.397861] [<ffffffff80266a00>] driver_probe_device+0x38/0x74
[    2.399806] [<ffffffff80266f70>] device_driver_attach+0x4c/0x50
[    2.401784] [<ffffffff80266fc8>] __driver_attach+0x54/0xc6
[    2.403810] [<ffffffff8026507e>] bus_for_each_dev+0x5a/0x98
[    2.405698] [<ffffffff80267054>] driver_attach+0x1a/0x22
[    2.407555] [<ffffffff8026581e>] bus_add_driver+0x12e/0x190
[    2.409417] [<ffffffff80267572>] driver_register+0x3a/0xd0
[    2.411331] [<ffffffff80268d32>] __platform_driver_register+0x1c/0x24
[    2.413384] [<ffffffff8041044e>] litex_soc_ctrl_driver_init+0x1a/0x22
[    2.415304] [<ffffffff80400cd8>] do_one_initcall+0x6e/0x134
[    2.417214] [<ffffffff80400f78>] kernel_init_freeable+0x172/0x1de
[    2.419164] [<ffffffff803b5018>] kernel_init+0x12/0xf0
[    2.421115] [<ffffffff8000302e>] ret_from_exception+0x0/0xc
[    2.426654] ---[ end trace c52f0c50dc8758d7 ]---
[    2.428148] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[    2.429979] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

gsomlo, 2021-06-26:
On Thu, Jun 24, 2021 at 04:32:06AM -0700, developandplay wrote:
> I revisited this issue and made some progress. Surprisingly I ran into a
> kernel panic.
> 
> [    2.362676] Oops - load access fault [#1]
> [    2.390956] Call Trace:
> [    2.391978] [<ffffffff802420a6>] litex_soc_ctrl_probe+0x3e/0x106
> [    2.394077] [<ffffffff80267f80>] platform_probe+0x3e/0x7c

Likely a CSR access in `litex_soc_ctrl_probe` triggering a cpu fault.
Could it be a lingering after-effect of fixing 64/32/16/8 bit access
strobes?

developandplay, 2021-06-27:
If you are referring to my access strobe fix for blackparrot then I don't think so.
I didn't touch anything Litex or Rocket related for that fix just Blackparrot's sytemverilog code.
Therefore Rocket simulation shouldn't be affected.

Unless you also ran into some strobe access problems on Rocket that I'm not aware of.

gsomlo, 2021-06-27:
On Sat, Jun 26, 2021 at 06:03:53PM -0700, developandplay wrote:
> If you are referring to my access strobe fix for blackparrot then I don't
> think so.

You're right, this is still using the RocketChip -- my mistake :)

The "load access fault" appears to happen here:

https://github.com/litex-hub/linux/blob/litex-rebase/drivers/soc/litex/litex_soc_ctrl.c#L45

... with the "bad address" being 0xffffffd004001b04 (presumably of the
32-bit scratch register, but that should be cross-referenced with the
csr map of the simulation).

This means CSR access is still somehow broken (in the simulator, since
it seems to work fine on a number of FPGAs)...

developandplay, 2021-06-27:
Is there some sort of (legitimate) address translation happening?
The scratch register should be at 0x12000004 according to the CSR map.

gsomlo, 2021-06-27:
On Sun, Jun 27, 2021 at 07:26:53AM -0700, developandplay wrote:
> Is there some sort of (legitimate) address translation happening?
> The scratch register should be at 0x12000004 according to the CSR map.

Yeah. On the ecpix5, csr_base is at 0x12000000 physical, but the value
for soc_ctl_dev->base returned by devm_platform_ioremap_resource() will
be 0xffffffd004016000, here: 

https://github.com/litex-hub/linux/blob/litex-rebase/drivers/soc/litex/litex_soc_ctrl.c#L102

developandplay, 2021-06-29:
Thanks for the EDIT now I have something to look into.
Up to this point I wasn't really sure what to look for.

OrkunAliOzkan, 2023-04-21:
> 

Hi, what exactly did you do in order to progressing on the 24th June, where you encountered a kernel panic?