litex-hub/linux-on-litex-rocket #19: sqrl_acorn support

sajattack, 2021-11-12:
I successfully got a basic thing running with a single core, no FPU, and BBL this morning. Now I'm trying to make a maxed out version with 4 cores, an fpu, and running at 100MHz. I'll post dts and bitstream and other bins here when I get it all working.

sajattack, 2021-11-12:
Basic config for now since I'm struggling to get crossover uart working on `--cpu-variant=full4d`
[acorn_cle_215.zip](https://github.com/litex-hub/linux-on-litex-rocket/files/7524762/acorn_cle_215.zip)

sajattack, 2021-11-23:
Here's everything I have for the full4d, including a boot.bin and initramfs for debian, dts, busybox boot.bin, bitstream, and driver folders. https://nextcloud.paulsajna.com/index.php/s/qi4oDSyXdSPQPE2

gsomlo, 2024-07-11:
closing, since sqrl acorn support was added with commit b9b92a5bf79334f100ad808df7827f1f326307c1

closed 2024-07-11 by gsomlo