litex-hub/linux-on-litex-rocket #20: is that possible to generate litex rocket with NVDLA?

12ff7a6, 2021-12-28:
Hello,
Chipyard support rocket with NVDLA, but this project only support VCU118. I want to implement this rocket with NVDLA project on other FPGA prototyping paltform. Which means I can't use Xilinx IPcores. So maybe litex open source IPs are good choice. Is that possible to generate litex rocket with NVDLA?

gsomlo, 2021-12-28:
On Tue, Dec 28, 2021 at 09:51:13AM -0800, 12ff7a6 wrote:
> Chipyard support rocket with NVDLA, but this project only support VCU118.

This project (a "computer" with a Rocket CPU and LiteX peripherals
around it) supports a whole bunch of Xilinx and Lattice/ECP5 based
development boards. I don't think the VCU118 is "officially" on that
list (yet), but if you can build LiteX bitstream using `--with-cpu rocket`
it should be possible to add support for it.

> I want to implement this rocket with NVDLA project on other FPGA prototyping
> paltform. Which means I can't use Xilinx IPcores. So maybe litex open source
> IPs are good choice. Is that possible to generate litex rocket with NVDLA?

That said, I don't know how NVDLA ties into it -- it's not a specific
development board, but rather some sort of ecosystem that one might be
able to run on different bits of hardware. I might be wrong about
that, this is the first time I've actually heard of NVDLA.

I'm only vaguely familiar with Chipyard, back when I was looking for a
solution it seemed to be analogous to LiteX (i.e., provide a
configurable set of peripherals one can place around Rocket to obtain
a "computer"), but at the time had a much more limited set of physical
development boards one could build it for, which did specifically
*not* include Lattice ECP5 based ones. This may have changed recently,
but I haven't kept up to date, so I really don't know what I'm talking
about :)

Anyway, long story short, I don't think I understand the question, please
clarify... :)

12ff7a6, 2021-12-28:
hi gsomlo, thank you for your reply.
let me clarify my problem.
NVDLA is short for Nvidia Deep Learning Accrelerator. [https://github.com/nvdla](url). Some SoC Designs has included this NVDLA such as Chipyard(Rocket core & NVDLA) [https://github.com/ucb-bar/chipyard](url) or ESP(Ariane & NVDLA) [https://github.com/sld-columbia/esp](url). Chipyard is using Chisel3 to generate RTL. I have already checked linux-on-litex-rocket and boot Rocket core on Arty A7 100T successfully, it seems linux-on-litex-rocket will first generate the RTLs(differnet configs) by using rocket-chip(so as Chipyard), then include those RTLs with some top levels. I wonder if I can plugin NVDLA while generate the RTLs? like modify the Chisel configs? 

Chipyard and ESP are all using Xilinx IPcores, such as DDR3/4, SGMII, PLL and so on. But I want to implement Rocket core with NVDLA on Cadance Protium prototyping platform, which means I do not use Xilinx workflow and I can't use Xilinx IPcores. 

I have 2 ideas, 

1. First generate RTLs by using Chipyard or ESP, then replace the Xilinx IPcores with Litex open source IPcores.
2. Generate rocket core and NVDLA RTL by using linux-on-litex-rocket workflow.

Which one do you think is reasonable?