litex-hub/linux-on-litex-rocket #21: Boot hang with DE2-115

lapnd, 2022-02-11:
Hello,
I'm trying with Altera DE2-115 kit. I have added ethernet patch to make TFTP boot work. However, the booting is hang at step `Freeing unused kernel image (initmem) memory: 2708K`
For the dts file, as the guide here https://github.com/litex-hub/linux-on-litex-rocket/issues/8
I made a copy of arty.dts, update few sessions
- Update the right address for UART
- Update RAM size
- Update session related to soc, ethernet, uart with output from litex_json2dts.py
[de2-115.dts.txt](https://github.com/litex-hub/linux-on-litex-rocket/files/8047953/de2-115.dts.txt)

This is the boot log. Could you please give me few hints on checking this issue?
Thank you!

```sh

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 11 2022 07:16:50
 BIOS CRC passed (608dacb6)

 Migen git sha1: ac70301
 LiteX git sha1: 7cc781f7

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
SDRAM:          65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x80000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x80000000 (Sequential, 2.0MiB)...
  Write speed: 5.8MiB/s
   Read speed: 9.7MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x80000000... (17824296 bytes)
Executing booted program at 0x80000000

--============= Liftoff! ===============--
bbl loader
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
[... 85 lines cut ...]
[    4.867794] TCP bind hash table entries: 512 (order: 1, 8192 bytes, linear)
[    4.878386] TCP: Hash tables configured (established 512 bind 512)
[    4.904076] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    4.915416] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    4.948946] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    8.140368] workingset: timestamp_bits=46 max_order=14 bucket_order=0
[   11.262946] LiteX SoC Controller driver initialized
[   22.015250] 12003000.serial: ttyLXU0 at MMIO 0x0 (irq = 0, base_baud = 0) is a liteuart
[   22.032188] printk: console [liteuart0] enabled
[   22.032188] printk: console [liteuart0] enabled
[   22.042772] printk: bootconsole [liteuart0] disabled
[   22.042772] printk: bootconsole [liteuart0] disabled
[   23.240688] loop: module loaded
[   25.878566] liteeth 12000800.mac: error -ENXIO: IRQ index 0 not found
[   25.887044] liteeth 12000800.mac: Failed to get IRQ, using polling
[   26.001220] liteeth 12000800.mac eth0: irq 0 slots: tx 2 rx 2 size 2048
[   26.254712] NET: Registered PF_INET6 protocol family
[   26.470926] Segment Routing with IPv6
[   26.480862] In-situ OAM (IOAM) with IPv6
[   26.497858] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   26.630914] NET: Registered PF_PACKET protocol family
[   26.893754] Warning: unable to open an initial console.
[   26.902926] remote fence extension is not available in SBI v0.1
[   27.446946] Freeing unused kernel image (initmem) memory: 2708K
```

gsomlo, 2022-02-11:
On Fri, Feb 11, 2022 at 02:48:40AM -0800, lapnd wrote:
> This is the boot log. Could you please give me few hints on checking this
> issue?

When you build your bitstream, does it pass timing?

Other than that, doublecheck the register addresses in DTS?

That's all I can think of, off the top of my head...

lapnd, 2022-02-12:
I can make it work now.
I'm not sure this is the root cause, but from the message `remote fence extension is not available in SBI v0.1`, I guest the issue may from SB1 v1.0. 
Enable this option on kernel make the linux boot ok
[image]
```sh
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Feb 12 2022 13:27:25
 BIOS CRC passed (54bd8546)

 Migen git sha1: ac70301
 LiteX git sha1: 7cc781f7

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
SDRAM:          65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x80000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x80000000 (Sequential, 2.0MiB)...
  Write speed: 5.8MiB/s
   Read speed: 9.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x80000000... (17824296 bytes)
Executing booted program at 0x80000000

--============= Liftoff! ===============--
bbl loader
              vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
                  vvvvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrrrr    vvvvvvvvvvvvvvvvvvvvvvvv
rrrrrrrrrrrrrrrr      vvvvvvvvvvvvvvvvvvvvvv  
rrrrrrrrrrrrr       vvvvvvvvvvvvvvvvvvvvvv    
rr                vvvvvvvvvvvvvvvvvvvvvv      
rr            vvvvvvvvvvvvvvvvvvvvvvvv      rr
rrrr      vvvvvvvvvvvvvvvvvvvvvvvvvv      rrrr
rrrrrr      vvvvvvvvvvvvvvvvvvvvvv      rrrrrr
rrrrrrrr      vvvvvvvvvvvvvvvvvv      rrrrrrrr
[... 93 lines cut ...]
[   39.050544] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   39.250182] NET: Registered PF_PACKET protocol family
[   39.763178] Warning: unable to open an initial console.
[   40.553822] Freeing unused kernel image (initmem) memory: 2708K
[   40.599206] Run /init as init process
[   40.603954]   with arguments:
[   40.610538]     /init
[   40.613552]   with environment:
[   40.620356]     HOME=/
[   40.623542]     TERM=linux
# ls
bin      etc      init     linuxrc  nfs      root     sys      usr
dev      home     lib      mnt      proc     sbin     tmp
# ifconfig eth0 192.168.1.50
# ping 192.168.1.100
192.168.1.100 is alive!
# ifconfig eth0
ifconfig: no support for status display
# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
# ping 192.168.1.100
192.168.1.100 is alive!
```

closed 2022-02-12 by lapnd

gsomlo, 2022-02-12:
On Sat, Feb 12, 2022 at 06:20:58AM -0800, lapnd wrote:
> I can make it work now.
> I'm not sure this is the root cause, but from the message remote fence
> extension is not available in SBI v0.1, I guest the issue may from SB1 v1.0.
> Enable this option on kernel make the linux boot ok

I'm not really sure how enabling RISCV_SBI_V01 in your kernel is in
any way related to the dev board, fpga model, or HDL compiler you're
using!

Seems like something about how you compiled the rest of your software
stack (i.e., busybox or whatever it is you're booting on top of
Linux).

Besides, the help line in the kernel Kconfig says RISCV_SBI_V01 is
being deprecated, so relying on it is not a good long-term strategy.

But good to hear it's working for you, at least! :)

developandplay, 2022-03-30:
Thanks @lapnd I recently tried recompiling the kernel and I ran into the same issue.
For me it would hang after `Timer extension is not available in SBI v0.1`.
Adding `CONFIG_RISCV_SBI_V01=y` to the defconfig like you suggested did the trick.