litex-hub/linux-on-litex-rocket #24: Simulator UART address

Kimplul, 2022-09-19:
Hi, apparently the UART in the simulator is placed at address `0x12003000` instead of `0x12002800` when run with `--with-sdram --sdram-init boot.bin`, copied from [here](https://github.com/litex-hub/linux-on-litex-rocket/issues/12#issuecomment-867562684). I'm not too familiar with the simulator, but there's no `main-ram` in my simulator, and if I try to add some with `--integrated-main-ram-size` I get the following error:

```
{'boot-sim1.bin': '00000000'}
INFO:SoC:        __   _ __      _  __
INFO:SoC:       / /  (_) /____ | |/_/
INFO:SoC:      / /__/ / __/ -_)>  <
INFO:SoC:     /____/_/\__/\__/_/|_|
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2022-09-19 15:40:33)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 1.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU rocket added.
INFO:SoC:CPU rocket adding IO Region 0 at 0x12000000 (Size: 0x70000000).
INFO:SoCRegion:Region size rounded internally from 0x70000000 to 0x80000000.
INFO:SoCBusHandler:io0 Region added at Origin: 0x12000000, Size: 0x70000000, Mode: RW, Cached: False Linker: False.
INFO:SoC:CPU rocket overriding rom mapping from 0x00000000 to 0x10000000.
INFO:SoC:CPU rocket overriding sram mapping from 0x01000000 to 0x11000000.
INFO:SoC:CPU rocket overriding main_ram mapping from 0x40000000 to 0x80000000.
INFO:SoC:CPU rocket setting reset address to 0x10000000.
INFO:SoC:CPU rocket adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 Bus adapted from Wishbone 64-bit to Wishbone 32-bit.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoC:CPU rocket adding Interrupt(s).
INFO:SoC:CPU rocket adding DMA Bus.
INFO:SoCDMABusHandler:Creating Bus Handler...
INFO:SoCDMABusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCDMABusHandler:Adding reserved Bus Regions...
INFO:SoCDMABusHandler:Bus Handler created.
INFO:SoCDMABusHandler:dma Region added at Origin: 0x00000000, Size: 0x100000000, Mode: RW, Cached: True Linker: False.
INFO:SoCDMABusHandler:dma added as Bus Slave.
INFO:SoCBusHandler:rom Region added at Origin: 0x10000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x10000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x11000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x11000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
ERROR:SoCBusHandler:main_ram Region in IO region, it can't be cached: Origin: 0x80000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False
ERROR:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x12000000, Size: 0x70000000, Mode: RW, Cached: False Linker: False
Bus Regions: (2)
rom                 : Origin: 0x10000000, Size: 0x00020000, Mode: RX, Cached: True Linker: False
sram                : Origin: 0x11000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False
Bus Masters: (1)
- cpu_bus0
Bus Slaves: (2)
- rom
- sram
```

Should the README be updated with the `sdram` flags (and `sim.dts` be modified) or am I doing something wrong here?

Here's my whole command line:
```
litex_sim --threads $(nproc) --with-ethernet --cpu-type rocket --cpu-variant linux --with-sdram --sdram-init boot.bin
```

gsomlo, 2022-10-10:
The long-term solution to this (and similar) issues will be to update the `litex/tools/litex_json2dts_linux.py` script to automatically generate the appropriate `.dts` file, and optionally call `dtc` on it and even more optionally build the resulting `.dtb` file into the bios blob included with the bitstream (or at least make it easyly available for loading into RAM as part of the sdcard/tftp/etc boot process).

At this time, the included `.dts` files are examples you can use to build upon to address your specific needs.

gsomlo, 2022-10-11:
On Mon, Sep 19, 2022 at 05:46:00AM -0700, Kimplul wrote:
> Hi, apparently the UART in the simulator is placed at address 0x12003000
> instead of 0x12002800 when run with --with-sdram --sdram-init boot.bin
> [...]
> Should the README be updated with the sdram flags (and sim.dts be modified) or
> am I doing something wrong here?

The actual placement of MMIO registers will inevitably change as
the LiteX source code evolves, so hard-coded values will be subject
to bit-rot and will need to be updated manually over time.

The long term solution is be to generate a .dts (and .dtb) file
automatically during build (see `litex/tools/litex_json2dts*.py`
for the current state of that effort).

gsomlo, 2024-07-11:
closing, please re-open if anything remains still un-addressed.

closed 2024-07-11 by gsomlo