litex-hub/linux-on-litex-rocket #25: Which cpu architecture to use in BBL building when full rocket (with fp) is used?

matsbror, 2022-12-05:
I am building the full cpu variant as I have the Kintex 7 device which can hold it, but I do not know which cpu architecture to specify when building BBL. Please advise.

gsomlo, 2022-12-05:
On Mon, Dec 05, 2022 at 03:48:00AM -0800, Mats Brorsson wrote:
> I am building the full cpu variant as I have the Kintex 7 device which can hold
> it, but I do not know which cpu architecture to specify when building BBL.
> Please advise.

You can use the same BBL parameters specified for the "linux"
variants. You get fpu emulation linked in, but if Linux never traps
into M mode due to a floating-point opcode, it doesn't matter.

matsbror, 2022-12-05:
Makes sense. Thanks!

closed 2022-12-05 by matsbror