litex-hub/linux-on-litex-rocket #26: Documentation of the varying rocket cpu-variants

matsbror, 2022-12-05:
Where can I find information about the different cpu-variants? E.g. linux4, full, linuxd?

gsomlo, 2022-12-05:
On Mon, Dec 05, 2022 at 04:10:03AM -0800, Mats Brorsson wrote:
> Where can I find information about the different cpu-variants? E.g. linux4,
> full, linuxd?

Look at https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/rocket/core.py#L45-L102

Also, the pre-built verilog is maintained in
https://github.com/litex-hub/pythondata-cpu-rocket. See
https://github.com/litex-hub/pythondata-cpu-rocket/blob/master/pythondata_cpu_rocket/verilog/update.sh
for how the pre-built variants are configured and built.

closed 2022-12-05 by matsbror

gsomlo, 2022-12-05:
On Mon, Dec 05, 2022 at 04:10:03AM -0800, Mats Brorsson wrote:
> Where can I find information about the different cpu-variants? E.g. linux4,
> full, linuxd?

The convention is [standard|linux|full][xCores][xMemBusWidth]

- linux supports S-mode and virtual memory
- full adds an FPU and Hypervisor support on top of that

There's an optional number of cores (e.g., 2, 4, 8).

Finally, there's [D|Q|O] for double (2x64 = 128), Quad (256), or
"Octo" (512) width of the cached-memory AXI interface between Rocket
and LiteX's LiteDRAM data port.

LiteX should warn you if your memory bus width is different from
LiteDRAM, which requires *additional* up- or down-conversion. Instead,
pick the appropriate variant to exactly match your board's LiteDRAM
port width.