litex-hub/linux-on-litex-rocket #29: Unable to boot Linux - userspace cannot write to console, no busybox shell

n-kremeris, 2023-03-22:
I'm building LiteX SoC with Rocket on a digilent nexys video board, which is listed as one of the supported platforms here.

gsomlo, 2023-03-22:
On Wed, Mar 22, 2023 at 08:45:41AM -0700, Norbert Kremeris wrote:
> I'm building LiteX SoC with Rocket on a digilent nexys video board, which is
> listed as one of the supported platforms here.

If you're using the pre-built binaries for software, chances are that
MMIO registers placement has changed in more recent gateware builds.

If you're putting together your own software, make sure your
devicetree .dts matches the way your gateware was built (use
`--csr-csv ./csr-csv` to ensure that information is captured during
build).

HTH,
--Gabriel

PS. please be warned that cpu variant naming is about to be changed
pretty significantly, which will make the
"linux-on-litex-rocket/README.md" (even more) obsolete, and in need of
an update (next on my list, I promise :) )

n-kremeris, 2023-03-22:
Hi Gabriel! Sorry, i accidentally posted the issue without writing all the info and then frantically started editing it. Didn't expect such a quick reply!

Please take a look again at the updated info when you have the time. I am indeed using the generated (but slightly modified) dts.

I am using my own build of linux, bbl, and tried a custom build of busybox/initramfs.cpio as well as the prebuilt one.

roryt12, 2023-03-22:
Sounds like the issues I had with the irq of liteuart? Take a look at https://github.com/roryt12/qmtech_wukong_debian_on_litex_rocket . It hanged exactly at this point if I enable irq vs pooling?

Also try an alternative , ie the hvc0 driver, at my try with naxvriscv https://github.com/roryt12/qmtech_wukong_debian_on_litex_rocket seems to work fine (needs more options in kernel and in dts )

gsomlo, 2023-03-22:
I haven't built anything using BBL in a very long time, having been discouraged from doing so by upstream who now favor
[opensbi](https://github.com/riscv-software-src/opensbi). The latter doesn't support emulating the FPU in M-mode, so it only works on boards that can accomodate at least one FPU-enabled Rocket core (85k ecp5, so no versa board; nexys* boards and better xilinx-equipped ones should still work).

I need to update the README to reflect this, but I'm also trying to get upstream LiteX to generate the full *.dts for a rocket-based system automatically, so I don't have to screw around with sample device tree files anymore :)

In the mean time, if you have a nexys_video, I'd like to propose that you try the following and let me know how it goes:

On the very latest upstream litex, add `#define CONFIG_BIOS_NO_BOOT` to the top of `litex/soc/software/bios/main.c` to inhibit it from automatically trying to boot from sdcard (you can still manually issue whatever boot command you want from the LiteX bios prompt). Then, build for the nexys_video:

```
litex-boards/litex_boards/targets/digilent_nexys_video.py --build \
    --cpu-type rocket --cpu-variant full --cpu-num-cores 2 --cpu-mem-width 2 --sys-clk-freq 50e6 \
    --with-ethernet --with-sdcard \
    --with-sata --sata-gen 1 --with-sata-pll-refclk \
    --csr-csv ./csr.csv
```

It's OK if you don't have a SATA adapter, but building it in anyway will make the MMIO registers line up properly with the DTS sample I'm about to link you to :)

Build the `litex-rebase` branch of the [`litex-hub` kernel tree](https://github.com/litex-hub/linux/tree/litex-rebase):

```
make clean
make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu-  litex_rocket_defconfig
make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- -j8
```

This gets you an `arch/riscv/boot/Image` kernel file.

Unpack [nexys_video_binaries.zip](https://github.com/litex-hub/linux-on-litex-rocket/files/11043565/nexys_video_binaries.zip). There will be a copy of `Image` already in there, pre-built. Feel free to use it or ignore it if you built your own. There's also a `satatest4imafdc.dts` sample in there. Compile it with:

```
dtc -O dtb satatest4imafdc.dts -o satatest4imafdc.dtb
```

Then build `fw_jump.bin` from [source](https://github.com/riscv-software-src/opensbi) (or use the included one), like so:

```
make CROSS_COMPILE=riscv64-unknown-linux-gnu- PLATFORM=generic \
    FW_FDT_PATH=~/...path/to/.../satatest2imafdcbkph.dtb FW_JUMP_FDT_ADDR=0x82400000
```

Use the included `initrd_bb` (which should not be significantly different from previous sets of instructions).

Place the included `boot.json`, `initrd_bb`, `Image`, and `fw_jump.bin` into a tftp server directory (or on your FAT formatted sdcard), and issue the appropriate boot command (`netboot` or `sdcardboot`) from the LiteX bios.

LMK if this works for you (it does for me, and it's how things will be done from now on -- no more BBL).

All of this is also perfectly capable to boot e.g. Fedora (with an appropriately modified DTS and some doctoring of the official raw image to get it installed to the sdcard :) ).

LMK how it goes -- good luck!

gsomlo, 2023-03-22:
note the *edited* `--cpu-variant linux --cpu-num-cores 4 --cpu-mem-width 2` command line. If you go for `full` (now with [H]ypervisor support), there's only room for 2 such cores, and I gave you the wrong DTS for that :)

gsomlo, 2023-03-22:
final edit: use `--cpu-type rocket --cpu-variant full --cpu-num-cores 2` to get H support, and be able to use the enclosed *binary* opensbi blob (updated the source DTS to reflect the two larger h-capable full cores vs. the previous 4 lighter `linux` ones). The updated/edited commend containing build instructions is now correct and should reference the correct zip file.

closed 2023-03-23 by n-kremeris