litex-hub/linux-on-litex-rocket #30: litex_sim fails loading opensbi: Liftoff hangs

OrkunAliOzkan, 2023-04-20:
Hi, I posted the issue on the main Litex repo (https://github.com/enjoy-digital/litex/issues/1672), but in retrospect heavily believe it to be much more appropriate to have here. I beg of you to review the URL listed above to see what it is I've been getting up to. I am unsure if copy pasting the entire message would be appropriate or not, but would be more than willing to if one were to think that I should.

gsomlo, 2023-04-20:
I did look through issue #1672. IMHO you're stretching the limits of
what is feasible to simulate with verilator :)

I do remember using simulation and verilator in the early days of
getting LiteX working with Rocket. But it was the low-end (mmu- and
fpu-less variant) I was using, and mostly just long enough to boot the
litex bios.

I remember trying to once boot linux in simulation, and it took about
8 hours of 8-core threaded verilator time to make it through the whole
boot process (and that was with bbl, so no actual experience booting
opensbi in simulation).

Generally, I get things hanging at liftoff mostly when the UART
register address is wrong in the .dts file (you need to get it right
both where the uart node is specified, and also on the kernel command
line in the "chosen" section).

So there's a small chance it's *not* related to the fact you're
simulating...

HTH, and good luck,
--Gabriel

OrkunAliOzkan, 2023-04-21:
Hello Gabriel!

To be honest hearing this all from you has been music to my ears, it has been disturbingly slow which was a bit disheartening. 

Right, back to the topic of conversation, I had checked all of which you said regarding the UART address in the .dts, and believe found it to be exactly as you specified, but I will upload it still as a point of reference, I would be very grateful if you could quickly just check through it to see if I have made any naive mistakes or errors. [dts_file_orkun.txt](https://github.com/litex-hub/linux-on-litex-rocket/files/11295102/dts_file_orkun.txt)

One thing I am quite keen to know is, by any chance is there a known method for which I could approximate the clock frequency of the SoC running the bootloader, either using the litex_sim application or through verilator itself? While having your info on the time taken to boot linux onto the simulated SoC using BBL is useful, it would be quite a valuable skill to know how to determine the time frame one could expect the simulator to complete a certain operation for any future projects, and also so I can know when I've waited for too long for the bootloader to do absolutely nothing at all.

Orkun :^]

OrkunAliOzkan, 2023-04-25:
Hi, removed any misc. questions I had in my original inquiry, apologies about that...

OrkunAliOzkan, 2023-04-26:
Hi I have resolved this issue now, I realised that if you want to have the correct address pin layout, well maybe use the litex_sim's in built csr generation tools (duh!)

I have a new issue however, the kernel hangs, but this is unrelated to this issues tab and if I must, I'll generate a new issue. I quite like the litex platform and I am quite keen to contribute by updating some of the documentation, and I would be absolutely delighted to contribute by adding more extensive documentation on how to boot linux on litex_sim using opensbi as the bootloader. Let me know if that is something desired

closed 2023-04-26 by OrkunAliOzkan

MooreNtu, 2025-01-09:
Hi, I have encountered the same issue with you, the liteX sim just hangs on liftoff , could you explain how to use litex_sim's build in csr generation tools?