litex-hub/linux-on-litex-rocket #34: Has anyone tried having this fabricated on the OpenMPW shuttle?

ryao, 2023-05-29:
Google pays to enable open source hardware to be fabricated for free: 

https://developers.google.com/silicon

Has anyone tried having this fabricated on that? If yes, how did it perform?

mithro, 2023-05-30:
FWIW - The Caravel Harness is based on the MCU version of LiteX / VexRISCV.

ryao, 2023-05-31:
> On Mon, May 29, 2023 at 03:43:16PM -0700, Richard Yao wrote: Has anyone tried having this fabricated on that? If yes, how did it perform?
> My $0.02: Rocket is the "tail wagging the dog" here -- the SoC's area will be overwhelmingly dominated by the Rocket CPU, with the rest of LiteX being rather small in comparison. So if you're interested in the idea of a "tapeout", the better question would be "has anyone done it for RocketChip yet?" Alternatively, consider the tapeout of LiteX with a *smaller* CPU, where the CPU wouldn't totally dominate the area/utilization of the resulting design. HTH, --Gabriel

I know someone using the Raspberry Pi CM4 to make devices, but has been looking for alternatives due to shortages. Seeing this made me think about whether it would be possible to produce an alternative for him through Google’s service by using this as a starting point and adding a display output for a touch screen, Ethernet, SPI, I2C, GPIO and eMMC. Unfortunately, LiteX does not provide a way to connect a LCD panel and get a frame buffer.

If there were a way to do that, the result would run a lightweight electron application (which would require porting electron to RISC-V) and need to have decent performance. The electron application is very light weight, as it is mostly HTML/CSS with a bit of JavaScript.

Getting back to my question, I asked mostly because there are two unknowns to me:

1. Whether it actually works when fabricated.
2. How it performs when fabricated. The demonstration video shows a particularly long boot time that would be problematic in a production device, so I assume that a tapeouf would be needed to raise the boot performance to something acceptable.

Having an open source SoC that can replace the CM4 for him is probably just a dream at this point due the need for a display output. People have implemented open source display outputs, but hooking them up to LiteX is a bit beyond my current ability:

https://github.com/hamsternz/DisplayPort_Verilog
https://github.com/skatanik/dsi_controller

Whether it would work well enough that you will not watch the screen repaint in slow motion when the web page in electron changes without GPU acceleration is also an unknown.

A GPU might be needed to accelerate drawing operations, but the open source GPU designs focus on GPGPU such that they do not support video outputs or have the Open GLES support needed to accelerate drawing operations in electron:

https://github.com/jbush001/NyuziProcessor
https://github.com/VerticalResearchGroup/miaow
https://github.com/vortexgpgpu/vortex
https://github.com/hughperkins/VeriGPU

Hooking that up to LiteX, a display output and Open GLES is definitely beyond my present ability, but I suppose that I can still dream.

gsomlo, 2023-05-31:
On Wed, May 31, 2023 at 12:40:15PM -0700, Richard Yao wrote:
> Getting back to my question, I asked mostly because there are two unknowns to
> me:
> 
>  1. Whether it actually works when fabricated.

No idea -- and IMHO there's really no good way to tell until someone
makes an attempt...

>  2. How it performs when fabricated. The demonstration video shows a
>     particularly long boot time that would be problematic in a production
>     device, so I assume that a tapeouf would be needed to raise the boot
>     performance to something acceptable.

The demo video (I assume you're taking about FOSDEM 23) shows it
booting Fedora on an ECP5 FPGA, with the system clock running at 50MHz
(fastest possible timing closure obtainable with yosys/trellis/nextpnr
on that particular design).

It is possible (likely?) that a well-done tapeout could get a
similarly configured SoC to run at higher frequencies (same design can
run at 100MHz on a Xilinx Kintex7 FPGA, and at 150MHz on a Virtex7,
so the 50MHz clock rate in the demo isn't an intrinsic property of the
SoC design :) ).

How much faster? -- Again, no idea until someone goes through the
exercise of doing at least a layout and simulation with the target PDK.
This in particular is where other examples of doing a rocketchip tapeout
might provide useful starting numbers...

Cheers,
--G

gsomlo, 2024-07-11:
closing since the conversation appears to have run its course :)

Feel free to follow-up or re-open if there's anything still left to discuss on this topic.

closed 2024-07-11 by gsomlo