litex-hub/linux-on-litex-rocket #38: build busybox : fatal error:crypt.h No such file

LYM-cat, 2023-11-12:
Hi,I I followed the readme to building busybox, but said I was missing this library <crypt.h>, what should I do？
[image]
[image]

LYM-cat, 2023-11-12:
I copy the <crypt.h> from /uer/include to /opt/riscv64-linux/sysroot/usr/include 。Then make it ,show   **undefined reference to `crypt'undefined reference to `crypt'**
 **undefined reference to `crypt'**
where should I wirte it?
[image]

gsomlo, 2023-11-13:
I think both the crypt.h header file and whatever library it
represents (probably libcrypt.so.*) should come as part of your
cross-compiler toolchain, e.g.
http://www.contrib.andrew.cmu.edu/~somlo/BTCP/RISCV-toolchain.tar.xz 

(see how to install it in the prerequisites section of the readme).

Copying from elsewhere on your computer to wherever the cross-compiled
busybox is looking for it is unlikely to be very useful.

HTH,
--Gabriel

gsomlo, 2024-07-11:
glibc removed crypt, so it's no longer part of the cross-compiler toolchain.

The new/updated (as of commit b9b92a5bf79334f100ad808df7827f1f326307c1) has an updated busybox config file that works around this issue.

Closing, please comment/reopen if that doesn't take care of it.

closed 2024-07-11 by gsomlo