litex-hub/linux-on-litex-rocket #44: support for nexys4ddr

juanschroeder, 2025-01-25:
Hi there,

I see the OpenSBI DTS for the nexys4ddr was renamed to digilent_nexys_video. 
I am trying to run Linux on linux-on-litex-rocket with a Nexys4DDR (Nexys A7) and up to the BIOS it works pretty well, but it it just hangs when trying to boot the binaries (I tried first serialboot and I am now trying network boot). 

I tried first with Nexys Video precompiled binaries and now I have tried rebuilding all binaries manually, using the previous nexys4ddr.dts that was in this repo before it was renamed. No success in either case.

Some questions are:
- Why was nexys4ddr.dts removed? What went wrong with the support for that board ?
- Would that old version of the DTS work with latest version of the repo if I build for Nexys4DDR?
- Any suggestion on how to debug this?

Thank you

gsomlo, 2025-01-25:
On Sat, Jan 25, 2025 at 01:00:50PM -0800, Juan Schroeder wrote:
> I see the OpenSBI DTS for the nexys4ddr was renamed to digilent_nexys_video
> not long ago.

The nexys4ddr and nexys-video are two different boards. The latter has
more RAM and (iirc) a larger version of the Artix7 FPGA.

I've been focusing on the ability to boot Fedora, and it didn't seem
like trying to pull it off within 256MB was worth the pain, so I
didn't test the latest project updates on that board.

> I am trying to run Linux on linux-on-litex-rocket with a Nexys4DDR
> (Nexys A7) and up to the BIOS it works pretty well, but it it just
> hangs when trying to boot the binaries (I tried first serialboot and
> I am now trying network boot).

That will happen because the DTS/DTB information in the nexys-video
precompiled binaries is blatantly wrong for the nexyx4ddr, so it will
try to access whatever registers it's looking for on the wrong address.

> I tried first with Nexys Video precompiled binaries and now I have tried
> rebuilding all binaries manually, using the previous nexys4ddr.dts that
> was in this repo before it was renamed. No success in either case.

You'd have to "reconcile" the nexys4ddr.dts with the current csr.csv
produced by building bitstream for the nexys4ddr (containing up to
date mmio register addresses for all relevant peripherals such as
uart, ethernet, etc.)

> Some questions are:
> 
>   • Why was nexys4ddr.dts removed? What went wrong with the support for that
>     board ?

see above -- 256MB RAM vs. trying to boot Fedora rv64 :)

>   • Would that old version of the DTS work with latest version of the repo
>     if I build for Nexys4DDR?

Yes, but you must update the old DTS with addresses and other
information from csr.csv, so that it matches the bitstream you're
building.

>   • Any suggestion on how to debug this?

If you get the uart right, it should produce output to the console,
which makes everything simpler from a debugging standpoint :)

HTH,
--Gabriel

juanschroeder, 2025-01-26:
Thanks for the fast reply.

I somehow made Linux boot in the Nexys A7 (A7-100T) with two cores and fixing the DTS as you suggested. To prevent kernel hanging on boot I had to disable this config (didn't check further):
CONFIG_RISCV_PROBE_UNALIGNED_ACCESS=n
CONFIG_BLK_DEV_NBD=n

I could see two cpus in /sysfs but only one in /proc/cpuinfo.
I also a few times had the Kernel hanging here in the following line (I didn't have time to check it yet). Does it ring a bell to you?
[    0.709624] smp: Bringing up secondary CPUs ...

I tried building the bitstream with 4 cores, or the 'full' variant of the CPU with just one core without success: it seems the A7-100T in this board doesn't have enough resources for that. I understand then that this project will be supporting boards with bigger FPGAs because of the Fedora focus. The LambdaConcept board is out of stock, so I guess currently my only choice would be to get a Nexys Video to try that. Would that FPGA be able to 'hold' more than two cores?

Thanks!

gsomlo, 2025-01-27:
On Sun, Jan 26, 2025 at 12:03:12PM -0800, Juan Schroeder wrote:
> I tried building the bitstream with 4 cores [...]

As of Litex commit c3e93620e (cca. Spring 2023), the variants have
been slightly renamed. If you want something capable of booting a
distro, you need `linux` or `full` (`linux` should be sufficient,
as it includes an FPU).

If you are content with being able to boot busybox on top of a linux
kernel, then `medium` (i.e., no FPU support) should work well enough.

I normally use 4 "linux" cores on the nexys-video. You should be able
to fit at least one of those on the nexys4ddr, maybe even 2.

> The LambdaConcept board is out of stock [...]

I've seen it go out of stock and then be available again a few times,
admittedly it's been a few years since I looked. I'd encourage you to
try to get in touch with someone at LambdaConcept and ask if they may
have plans to do another production run any time soon.

The advantage of an ECP5 Lattice FPGA is that it can be programmed
with a 100% Free (as in Freedom) toolchain (yosys/trellis/nextpnr).

The downside is that one can only fit a single `linux` variant rocket
core (enough to run Fedora, but only barely :)

HTH,
--Gabriel